sijones/DiyBatteryBMS #4: pins_arduino.h: No such file or directory

robwolff3, 2024-06-28:
Hey sijones, thank you so much for sharing this tool, its exactly what I needed for my diy battery setup! I'm running into the following error when trying to run the first command `PlatformIO: Upload([device chosen])`. Do you happen to know whats going on here?

`esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory`

```
*  Executing task: C:\Users\Rob\.platformio\penv\Scripts\platformio.exe run --target upload --environment esp32dev 

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.7.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- AsyncMqttClient-esphome @ 2.1.0
|-- ArduinoJson @ 7.1.0+sha.22dd4da
|-- Time @ 1.6.1+sha.a18e50d
|-- mcp_can @ 1.5.0+sha.5439f06
|-- AsyncElegantOTA @ 2.2.8+sha.0d7e6f1
|-- FS @ 2.0.0
|-- LittleFS @ 2.0.0
|-- WiFi @ 2.0.0
|-- SPI @ 2.0.0
|-- Preferences @ 2.0.0
|-- WebServer @ 2.0.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
Building in debug mode
Compiling .pio\build\esp32dev\src\CANBUS.cpp.o
Compiling .pio\build\esp32dev\src\VEDisplay.cpp.o
Compiling .pio\build\esp32dev\src\VeDirectFrameHandler.cpp.o
Compiling .pio\build\esp32dev\src\WifiMQTTManager.cpp.o
Compiling .pio\build\esp32dev\src\mEEPROM.cpp.o
Compiling .pio\build\esp32dev\src\main.cpp.o
Building .pio\build\esp32dev\bootloader.bin
Generating partitions .pio\build\esp32dev\partitions.bin
In file included from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from src/CANBUS.h:13,
                 from src/CANBUS.cpp:1:
C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/libraries/Preferences/src/Preferences.h:17,
                 from src/mEEPROM.h:7,
                 from src/mEEPROM.cpp:1:
C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
In file included from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from src/VeDirectFrameHandler.cpp:36:
C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
In file included from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from src/WifiMQTTManager.h:3,
                 from src/WifiMQTTManager.cpp:3:
C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\src\CANBUS.cpp.o] Error 1
*** [.pio\build\esp32dev\src\mEEPROM.cpp.o] Error 1
In file included from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from src/VEDisplay.cpp:1:
C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\src\VeDirectFrameHandler.cpp.o] Error 1
*** [.pio\build\esp32dev\src\WifiMQTTManager.cpp.o] Error 1
*** [.pio\build\esp32dev\src\VEDisplay.cpp.o] Error 1
In file included from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from src/main.cpp:22:
C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for esptool.py v4.5.1
pins_arduino.hCreating esp32 image...
 dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > Merged 1 ELF section
https://registry.platformio.org/search?q=header:pins_arduino.hSuccessfully created esp32 image.

*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
======================================================== [FAILED] Took 10.85 seconds ========================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:10.849
=================================================== 1 failed, 0 succeeded in 00:00:10.849 ===================================================

 *  The terminal process "C:\Users\Rob\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'esp32dev'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

sijones, 2024-06-28:
Can you try cleaning the project (Bin icon at bottom of VS Code), then delete the contents of .pio\libdeps\esp32dev

Then instead of Upload, choose Build.

Also try choosing the esp32dev environment, at the bottom it probably is on default, choose esp32dev.

robwolff3, 2024-06-28:
Project cleaned, `.pio\libdeps\esp32dev` deleted and esp32dev is selected as the environment. Getting the same error unfortunately.

```
 *  Executing task: C:\Users\Rob\.platformio\penv\Scripts\platformio.exe run --environment esp32dev 

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ ^2.1.3
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.1.3 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ ^3.2.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.2.2 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing heman/AsyncMqttClient-esphome @ ^2.1.0
Unpacking  [####################################]  100%
Library Manager: AsyncMqttClient-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing git+https://github.com/bblanchon/ArduinoJson.git
git version 2.45.2.windows.1
Cloning into 'C:\Users\Rob\.platformio\.cache\tmp\pkg-installing-jhwigarr'...
remote: Enumerating objects: 594, done.
remote: Counting objects: 100% (594/594), done.
remote: Compressing objects: 100% (467/467), done.
remote: Total 594 (delta 117), reused 206 (delta 61), pack-reused 0
Receiving objects: 100% (594/594), 388.05 KiB | 4.67 MiB/s, done.
Resolving deltas: 100% (117/117), done.
Library Manager: ArduinoJson@7.1.0+sha.22dd4da has been installed!
Library Manager: Installing git+https://github.com/PaulStoffregen/Time.git
git version 2.45.2.windows.1
Cloning into 'C:\Users\Rob\.platformio\.cache\tmp\pkg-installing-104y_5bz'...
remote: Enumerating objects: 38, done.
remote: Counting objects: 100% (38/38), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 38 (delta 2), reused 25 (delta 0), pack-reused 0
Receiving objects: 100% (38/38), 26.01 KiB | 2.17 MiB/s, done.
Resolving deltas: 100% (2/2), done.
Library Manager: Time@1.6.1+sha.a18e50d has been installed!
Library Manager: Installing git+https://github.com/coryjfowler/MCP_CAN_lib.git#1.5.0
git version 2.45.2.windows.1
Cloning into 'C:\Users\Rob\.platformio\.cache\tmp\pkg-installing-4xt34o9h'...
remote: Enumerating objects: 33, done.
remote: Counting objects: 100% (33/33), done.
remote: Compressing objects: 100% (26/26), done.
remote: Total 33 (delta 2), reused 19 (delta 0), pack-reused 0
Receiving objects: 100% (33/33), 42.17 KiB | 1.20 MiB/s, done.
Resolving deltas: 100% (2/2), done.
Note: switching to '5439f064f4e6e06265407807bf87df401442cc3a'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Library Manager: mcp_can@1.5.0+sha.5439f06 has been installed!
Library Manager: Installing git+https://github.com/ayushsharma82/AsyncElegantOTA.git
git version 2.45.2.windows.1
Cloning into 'C:\Users\Rob\.platformio\.cache\tmp\pkg-installing-95dfo8mn'...
remote: Enumerating objects: 30, done.
remote: Counting objects: 100% (30/30), done.
[... 132 lines cut ...]
                 from src/main.cpp:22:
C:/Users/Rob/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
======================================================== [FAILED] Took 28.42 seconds ========================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:28.417
=================================================== 1 failed, 0 succeeded in 00:00:28.417 ===================================================

 *  The terminal process "C:\Users\Rob\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32dev'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

sijones, 2024-06-28:
I'd check you've got plenty of space, it fails in a different place in that second log.

Maybe also try making a folder on C:\ like C:\source\diy-battery 

I've tried replicating it and it's working, is any other project compiling for you?

There is a lib update, so alter the platformio file and change
esphome/ESPAsyncWebServer-esphome@^3.2.0
to
esphome/ESPAsyncWebServer-esphome@^3.2.2

I doubt that's the issue tho.

sijones, 2024-06-28:
just to clarify, I've tried replicating it and can't, it's working for me.

If you can't get it working then i'll look to do a workaround, just need to put a change in the software to support it tho.

sijones, 2024-06-28:
I don't have a mac so haven't been able to see the problem there, did know it didn't compile and it's to do with what you've found, i changed the directory from .\src to ./src to try and see if that helped with mac, obviously hasn't fixed it, let me know what the workaround you've done.

robwolff3, 2024-06-28:
I changed it from `.\src` to just `src`. Not sure if that screws it up as a relative path. Just a quick bandaid to get it to build.

closed 2024-06-29 by sijones