sijones/DiyBatteryBMS #7: Build for ESP32-S3-DevKitC-1 boards

NNenya, 2024-08-16:
Any chance of there being a build for the ESP32-S3-DevKitC-1 board?

I recompiled your code for the ESP32-S3-DevKitC-1 environment in VSCode but the wifi (ap and station) was cycling repeatedly start to stop. This prevented any connections to the ap, or from the station (esp32s3) to my wifi router when the ssid and password was manually added to WifiMQTTManager.h.

sijones, 2024-08-16:
Am currently away, but will look when am back.

Did you alter the platformio file for your platform? Or just use the generic esp32 ?

NNenya, 2024-08-16:
Much appreciated. 

Yes, I added the below to platformio.ini and was able to boot after upload, but immediately went into an endless cycle.

[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
board_build.filesystem = littlefs
framework = arduino
lib_deps = 
    ;Time
	esphome/AsyncTCP-esphome@^2.1.3
	esphome/ESPAsyncWebServer-esphome@^3.2.2
	heman/AsyncMqttClient-esphome@^2.1.0
	https://github.com/bblanchon/ArduinoJson.git
	https://github.com/PaulStoffregen/Time.git
	https://github.com/coryjfowler/MCP_CAN_lib.git#1.5.0
	https://github.com/sijones/AsyncElegantOTA.git
	https://github.com/hasenradball/LCD-I2C.git
upload_protocol = esptool
monitor_speed = 115200
;upload_port = COM3
;debug_port = COM6
board_build.partitions = default.csv
build_type = debug
build_flags = -DCORE_DEBUG_LEVEL=4 -DCAN_BUS_CS_PIN=2 -DVEDIRECT_RX=33 -DVEDIRECT_TX=32 -DONEWIREPIN=12 -DCAN0_INT=22

sijones, 2024-08-19:
I've bought an S3 board, once it arrives I'll take a look at what needs to be changed to support it.

sijones, 2024-08-20:
It's fixed, if you resync your main branch and try build / upload etc, please make sure you flash erase your esp before then uploading this new build, this is because the serial pins have changed and you'll have the old ones saved to flash so you won't see it work for you.

NNenya, 2024-08-20:
Many thanks for this, but the once added to my network the webpage doesn't load, although I can ping it reliably.

sijones, 2024-08-20:
You need to flash the second part, in the wiki you'll see it says to "Flash Filesystem" after uploading the firmware.

NNenya, 2024-08-20:
I did.

[image]

sijones, 2024-08-20:
What does the serial port output say, it should look similar to this below which is same device as yourself, this is the boot after initial configuration to wifi network:

> --- Terminal on COM8 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

>ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   247][D][main.cpp:105] setup(): EEPROM Store opened, initial key found.
[   259][D][WifiMQTTManager.cpp:7] begin(): Getting all values from EEPROM
[   267][D][WifiMQTTManager.cpp:12] begin(): Wifi SSID: jameater, Password: hetty1980
[   284][D][WifiMQTTManager.cpp:47] begin(): Connecting to SSID: jameater
[   306][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   348][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[   366][D][WifiMQTTManager.cpp:52] begin(): Setting up mDNS Service.
[   386][D][WifiMQTTManager.cpp:86] begin(): MQTT details stored are not valid.
[   393][D][WifiMQTTManager.cpp:87] begin(): MQTT Server IP: , MQTT Port 1883, MQTT Client ID: `��?␟
Entering Configuration Mode Failure...
[   610][E][CANBUS.cpp:57] Begin(): CAN Bus Failed to Initialise
[  1376][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  1384][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 39 - TIMEOUT
[  1391][D][WiFiGeneric.cpp:1082] _eventCallback(): WiFi Reconnect Running
[  1406][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 5
[  1412][D][mqttFunctions.h:107] WiFiEvent(): WiFi lost connection
[  1474][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  1482][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 4
[  1490][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  1498][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 10.10.10.168, MASK: 255.255.255.0, GW: 10.10.10.1
[  1508][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 7
[  1514][D][mqttFunctions.h:103] WiFiEvent(): WiFi connected, IP Address: 10.10.10.168
[  1518][D][HTTPWSFunctions.h:9] setClock(): Setting Clock
[  5027][D][HTTPWSFunctions.h:18] setClock(): NTP time Tue Aug 20 22:06:55 2024
[  5035][D][HTTPWSFunctions.h:484] taskStartWebServices(): Starting Web Services
[  5042][D][HTTPWSFunctions.h:402] StartCriticalWebService(): Configuring Web Services.
[  5050][D][HTTPWSFunctions.h:416] StartCriticalWebService(): Redirecting root to index.htm
[ 15620][D][main.cpp:222] loop(): Send MQTT Param Update
[ 15628][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico does not exist, no permits for creation
[ 15639][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico.gz does not exist, no permits for creation
[ 15650][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico/index.htm does not exist, no permits for creation
[ 15662][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico/index.htm.gz does not exist, no permits for creation
[ 15674][I][HTTPWSFunctions.h:371] onEvent(): ws[/ws][1] connected
[ 15766][I][HTTPWSFunctions.h:382] onEvent(): ws[/ws][1] pong[0]:

sijones, 2024-08-20:
What does the serial port output say, it should look similar to this below which is same device as yourself, this is the boot after initial configuration to wifi network:

> --- Terminal on COM8 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

>ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   247][D][main.cpp:105] setup(): EEPROM Store opened, initial key found.
[   259][D][WifiMQTTManager.cpp:7] begin(): Getting all values from EEPROM
[   267][D][WifiMQTTManager.cpp:12] begin(): Wifi SSID: *****, Password: *****
[   284][D][WifiMQTTManager.cpp:47] begin(): Connecting to SSID: jameater
[   306][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   348][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[   366][D][WifiMQTTManager.cpp:52] begin(): Setting up mDNS Service.
[   386][D][WifiMQTTManager.cpp:86] begin(): MQTT details stored are not valid.
[   393][D][WifiMQTTManager.cpp:87] begin(): MQTT Server IP: , MQTT Port 1883, MQTT Client ID: `��?␟
Entering Configuration Mode Failure...
[   610][E][CANBUS.cpp:57] Begin(): CAN Bus Failed to Initialise
[  1376][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  1384][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 39 - TIMEOUT
[  1391][D][WiFiGeneric.cpp:1082] _eventCallback(): WiFi Reconnect Running
[  1406][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 5
[  1412][D][mqttFunctions.h:107] WiFiEvent(): WiFi lost connection
[  1474][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  1482][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 4
[  1490][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  1498][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 10.10.10.168, MASK: 255.255.255.0, GW: 10.10.10.1
[  1508][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 7
[  1514][D][mqttFunctions.h:103] WiFiEvent(): WiFi connected, IP Address: 10.10.10.168
[  1518][D][HTTPWSFunctions.h:9] setClock(): Setting Clock
[  5027][D][HTTPWSFunctions.h:18] setClock(): NTP time Tue Aug 20 22:06:55 2024
[  5035][D][HTTPWSFunctions.h:484] taskStartWebServices(): Starting Web Services
[  5042][D][HTTPWSFunctions.h:402] StartCriticalWebService(): Configuring Web Services.
[  5050][D][HTTPWSFunctions.h:416] StartCriticalWebService(): Redirecting root to index.htm
[ 15620][D][main.cpp:222] loop(): Send MQTT Param Update
[ 15628][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico does not exist, no permits for creation
[ 15639][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico.gz does not exist, no permits for creation
[ 15650][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico/index.htm does not exist, no permits for creation
[ 15662][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico/index.htm.gz does not exist, no permits for creation
[ 15674][I][HTTPWSFunctions.h:371] onEvent(): ws[/ws][1] connected
[ 15766][I][HTTPWSFunctions.h:382] onEvent(): ws[/ws][1] pong[0]:

NNenya, 2024-08-20:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377560
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   245][D][main.cpp:105] setup(): EEPROM Store opened, initial key found.
[   257][D][WifiMQTTManager.cpp:7] begin(): Getting all values from EEPROM
[   265][D][WifiMQTTManager.cpp:12] begin(): Wifi SSID: XXXXX, Password: xxxxxxxx
[   282][D][WifiMQTTManager.cpp:47] begin(): Connecting to SSID: XXXXX
[   297][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   340][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[   376][D][WifiMQTTManager.cpp:52] begin(): Setting up mDNS Service.
[   706][D][mqttFunctions.h:213] mqttsetup(): Using IP Address method to connect.
Entering Configuration Mode Failure...
[   920][E][CANBUS.cpp:57] Begin(): CAN Bus Failed to Initialise
[  1177][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  1185][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 4

[  2688][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  2696][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 192.168.1.102, MASK: 255.255.255.0, GW: 192.168.1.1
[  2706][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 7

[  2712][D][mqttFunctions.h:103] WiFiEvent(): WiFi connected, IP Address: ␐  ?␏
[  2720][I][mqttFunctions.h:91] connectToMqtt(): Connecting to MQTT...
[  2726][I][mqttFunctions.h:93] connectToMqtt(): Using User: mqtt, Password: mqtt
[  2728][D][HTTPWSFunctions.h:9] setClock(): Setting Clock
[  2749][D][mqttFunctions.h:114] onMqttConnect(): Connected to MQTT.
[  2766][D][mqttFunctions.h:149] onMqttMessage(): Message received. Topic: DIYBATTERYBMS/set/ChargeEnable, Payload: ONhargeenabled":true,"forcecharge":false,"chargecurrent":0,"dischargecurrent":0,"totalheap":348572,"freeheap": ␘ ?␇␃
[ 15930][D][main.cpp:222] loop(): Send MQTT Param Update
[ 30936][D][main.cpp:222] loop(): Send MQTT Param Update
[ 45942][D][main.cpp:222] loop(): Send MQTT Param Update
[ 60948][D][main.cpp:222] loop(): Send MQTT Param Update
[ 75954][D][main.cpp:222] loop(): Send MQTT Param Update
[ 90960][D][main.cpp:222] loop(): Send MQTT Param Update
[105966][D][main.cpp:222] loop(): Send MQTT Param Update

NNenya, 2024-08-20:
Here is serial output from another S3 device:

[18:26:54]ESP-ROM:esp32s3-20210327
[18:26:54]Build:Mar 27 2021
[18:26:54]rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
[18:26:54]Saved PC:0x420b572e
[18:26:54]SPIWP:0xee
[18:26:54]mode:DIO, clock div:1
[18:26:54]load:0x3fce3808,len:0x4bc
[18:26:54]load:0x403c9700,len:0xbd8
[18:26:54]load:0x403cc700,len:0x2a0c
[18:26:54]entry 0x403c98d0
[18:26:55][   249][D][main.cpp:105] setup(): EEPROM Store opened, initial key found.
[18:26:55][   262][D][WifiMQTTManager.cpp:7] begin(): Getting all values from EEPROM
[18:26:55][   270][D][WifiMQTTManager.cpp:12] begin(): Wifi SSID: XXXXX, Password: xxxxxxxx
[18:26:55][   279][E][Preferences.cpp:483] getString(): nvs_get_str len fail: MQTTServerIP NOT_FOUND
[18:26:55][   295][D][WifiMQTTManager.cpp:47] begin(): Connecting to SSID: XXXXX
[18:26:55][   316][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[18:26:55][   358][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[18:26:55][   378][D][WifiMQTTManager.cpp:52] begin(): Setting up mDNS Service.
[18:26:55][   707][D][WifiMQTTManager.cpp:92] begin(): MQTT server not set.
[18:26:55][   713][D][WifiMQTTManager.cpp:93] begin(): MQTT Server IP: , MQTT Port 1883, MQTT Client ID: \��?
[18:26:55][   930][E][CANBUS.cpp:57] Begin(): CAN Bus Failed to Initialise
[18:26:55]E (2053) wifi:Set status to INIT
[18:26:55][  1136][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[18:26:55][  1146][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 203 - ASSOC_FAIL
[18:26:55][  1152][D][WiFiGeneric.cpp:1082] _eventCallback(): WiFi Reconnect Running
[18:26:55][  1167][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 5
[18:26:55]
[18:26:55][  1174][D][mqttFunctions.h:107] WiFiEvent(): WiFi lost connection
[18:26:56]E (2847) wifi:Set status to INIT
[18:26:56][  1930][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[18:26:56][  1940][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 203 - ASSOC_FAIL
[18:26:56][  1947][D][WiFiGeneric.cpp:1086] _eventCallback(): WiFi AutoReconnect Running
[18:26:56][  1962][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 5
[18:26:56]
[18:26:56][  1968][D][mqttFunctions.h:107] WiFiEvent(): WiFi lost connection
[18:26:57][  2843][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[18:26:57][  2851][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 4
[18:26:57]
[18:26:57][  2857][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[18:26:57][  2864][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 192.168.1.146, MASK: 255.255.255.0, GW: 192.168.1.1
[18:26:57][  2875][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 7
[18:26:57]
[18:26:57][  2881][D][mqttFunctions.h:103] WiFiEvent(): WiFi connected, IP Address: 0��?
[18:26:57][  2938][D][HTTPWSFunctions.h:9] setClock(): Setting Clock
[18:27:10][ 15940][D][main.cpp:222] loop(): Send MQTT Param Update
[18:27:25][ 30946][D][main.cpp:222] loop(): Send MQTT Param Update

sijones, 2024-08-20:
Does the wifi AP that you're connecting to allow internet, looks to me like it's stuck in the loop waiting for the Clock to be set before then loading the web services.

NNenya, 2024-08-20:
It doesn't allow internet. Is access to the internet a must?

NNenya, 2024-08-20:
Yep, that's the reason. Any chance this can be pointed to an internal NTP server?

sijones, 2024-08-20:
Right now it will fail as the GUI uses CDN for the html framework to be downloaded (this speeds up loading time as the esp32 doesn't have to serve it) and the clock function.

So if you want to use it you'll need to allow access to the NTP servers:
0.uk.pool.ntp.org
1.uk.pool.ntp.org

and CDN:
https://cdn.jsdelivr.net 

I'll have a think about a workaround and see what I can do for another release, will be a few days.

NNenya, 2024-08-20:
I've opened it up to those hosts. I understand the convenience of internet resources but eliminating that dependence would be perfect for a battery monitor.

Many thanks for your help getting this far, now on the next phase, canbus comms to xanbus.

closed 2024-08-20 by NNenya

sijones, 2024-08-24:
The current main branch now has configurable NTP server support, you can specify one or two by comma delimiting the input of the host names. The web browser accessing this will need internet access as the webpage has the url to download the framework from the CDN, you don't need the device to have access to the CDN, that was a bit of a mistake telling you that.

A new release will be done shortly, just finishing and testing another feature first.