sijones/DiyBatteryBMS #8: BMS parameters not saving to EEPROM on ESP32-S3-DevKitC-1 board

NNenya, 2024-08-24:
Parameters are not surviving reboot/reset. See serial monitor below ....

-------
 *  Executing task: ~\.platformio\penv\Scripts\platformio.exe device monitor --environment esp32-s3-devkitc-1 --port COM15 
--- Terminal on COM15 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[ 22853][I][HTTPWSFunctions.h:427] onEvent(): ws[/ws][1] connected

[ 22873][I][HTTPWSFunctions.h:438] onEvent(): ws[/ws][1] pong[0]: 

[ 22879][I][HTTPWSFunctions.h:439] onEvent(): Sending All Data to All WS Clients
[ 34211][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[27]: 
[ 34218][D][HTTPWSFunctions.h:449] onEvent(): {"chargevoltage" : "55600"}

[ 38647][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[30]: 
[ 38654][D][HTTPWSFunctions.h:449] onEvent(): {"dischargevoltage" : "45000"}

[ 43332][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[25]: 
[ 43339][D][HTTPWSFunctions.h:449] onEvent(): {"fullvoltage" : "55200"}

[ 54369][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[31]: 
[ 54376][D][HTTPWSFunctions.h:449] onEvent(): {"maxchargecurrent" : "100000"}

[ 65729][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[34]: 
[ 65736][D][HTTPWSFunctions.h:449] onEvent(): {"maxdischargecurrent" : "100000"}

[ 75554][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[30]: 
[ 75560][D][HTTPWSFunctions.h:449] onEvent(): {"batterycapacity" : "460000"}

[ 79646][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[22]: 
[ 79653][D][HTTPWSFunctions.h:449] onEvent(): {"lowsoclimit" : "10"}

[ 80569][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[24]: 
[ 80576][D][HTTPWSFunctions.h:449] onEvent(): {"highsoclimit" : "100"}

[ 83984][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[25]: 
[ 83991][D][HTTPWSFunctions.h:449] onEvent(): {"slowchargesoc1" : "90"}

[ 87143][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[27]: 
[ 87149][D][HTTPWSFunctions.h:449] onEvent(): {"slowchargesoc1div" : "9"}

[ 89367][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[25]: 
[ 89374][D][HTTPWSFunctions.h:449] onEvent(): {"slowchargesoc2" : "95"}

[ 92252][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[28]: 
[ 92259][D][HTTPWSFunctions.h:449] onEvent(): {"slowchargesoc2div" : "14"}

[ 93564][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[18]: 
[ 93571][D][HTTPWSFunctions.h:449] onEvent(): {"saveall" : true}

[114655][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[27]: 
[114662][D][HTTPWSFunctions.h:449] onEvent(): {"pylontechenabled" : true}

[119865][D][HTTPWSFunctions.h:446] onEvent(): ws[/ws][1] text-message[17]: 
[119872][D][HTTPWSFunctions.h:449] onEvent(): {"reboot" : true}

[119905][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[119913][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[119920][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 5

[119927][D][mqttFunctions.h:107] WiFiEvent(): WiFi lost connection
[119933][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 3 - STA_STOP
[119940][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 3

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377560
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   250][D][main.cpp:105] setup(): EEPROM Store opened, initial key found.
[   262][D][WifiMQTTManager.cpp:7] begin(): Getting all values from EEPROM
[   270][D][WifiMQTTManager.cpp:12] begin(): Wifi SSID: XXXXX, Password: xxxxxxxx
[   287][D][WifiMQTTManager.cpp:47] begin(): Connecting to SSID: XXXXX
[   308][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   350][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[   369][D][WifiMQTTManager.cpp:52] begin(): Setting up mDNS Service.
[   389][D][mqttFunctions.h:213] mqttsetup(): Using IP Address method to connect.
[   603][E][CANBUS.cpp:57] Begin(): CAN Bus Failed to Initialise
[   613][D][HTTPWSFunctions.h:9] TaskSetClock(): Entering TaskSetClock
[  1239][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  1246][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 4

[  1450][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  1458][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 192.168.21.141, MASK: 255.255.255.0, GW: 192.168.21.1
[  1468][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 7

[  1474][D][mqttFunctions.h:103] WiFiEvent(): WiFi connected, IP Address: l  ?␏
[  1482][I][mqttFunctions.h:91] connectToMqtt(): Connecting to MQTT...
[  1488][I][mqttFunctions.h:93] connectToMqtt(): Using User: xxxxx, Password: xxxxx
[  1512][D][HTTPWSFunctions.h:539] taskStartWebServices(): Starting Web Services
[  1620][D][HTTPWSFnctions.h:39] TaskSetClock(): SettingClock
[  1735][D][HTTPWSFunctions.h:57] TaskSetClock(): NTP time Sat Aug 24 19:39:17 2024

[  4102][I][HTTPWSFunctions.h:427] onEvent(): ws[/ws][1] connected

[  4114][I][HTTPWSFunctions.h:438] onEvent(): ws[/ws][1] pong[0]:

[  4121][I][HTTPWSFunctions.h:439] onEvent(): Sending All Data to All WS Clients
-------

The webui afterwards shows unchanged parameters...

[image]
[image]

sijones, 2024-08-24:
Thanks, confirmed I can reproduce this so will try and fix asap.

sijones, 2024-08-24:
Looks like a a EEPROM Lib issue on the S3 platform.