sijones/DiyBatteryBMS #10: Endless reboot loop when connecting to ESP access point

NeliusDK, 2024-09-28:
I've tried getting the ESP32-S3 working, no luck at all, access point newer gets activated after flash.
Seems like nothing is being loaded.

As i can see only the old ESP32 board linked, i then tried flashing 2 different older ESP-32-wroom-32 boards, getting a little longer, but both of them having the same new problem, with an endless reboot as seconds after the DIYBATTERYBMS access point gets activated.
I can succesfully do the OTA update with littlefs.bin.
The access point gets activated again, but its not possible to select a network and enter password, before the ESP reboots and you need to start over.

NeliusDK, 2024-09-28:
Not sure if relevant, but i used the "ESP Flash Tool Method" NOT VS Code.

sijones, 2024-09-28:
Generally when it reboots constantly during boot it's because the allocated Pins are in use by a built in device, like PSRAM, and other things.

If you use Putty and open the serial port on the device, that should give a debug output, grab that and let me see where the reboot point is.

NeliusDK, 2024-09-28:
`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    10][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, A                                                       PB: 80000000 Hz
[    32][D][main.cpp:81] setup(): EEPROM not setup, writing inital values.
[    41][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'CAN_CS_PIN' = '5'
[    48][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'VictronRX' = '33'
[    55][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'VictronTX' = '32'
[    62][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'ChargeVolt' = '55200'
[    69][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'ChargeCurr' = '0'
[    76][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'DischargeVolt' = '45000'
[    84][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'DischargeCurr' = '0'
[    91][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'LowSOCLimit' = '0'
[    98][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'HighSOCLimit' = '100'
[   105][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'BattCapacity' = '0'
[   113][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'VE_WAIT_TIME' = '1'
[   120][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'VE_STARTUP_TIME' = '30'
[   127][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'VE_LCD_REFRESH' = '1'
[   135][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'VE_MQTT_REC' = '5'
[   142][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'VE_LOOP_TIME' = '5'
E (144) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1367:error: C                                                       orrupted dir pair at {0x0, 0x1}

E (145) esp_littlefs: mount failed,  (-84)
E (148) esp_littlefs: Failed to initialize LittleFS
[   160][E][LittleFS.cpp:98] begin(): Mounting LittleFS failed! Error: -1
[   178][D][main.cpp:132] setup(): Failed to mount storage (LittleFS).
[   184][D][WifiMQTTManager.cpp:7] begin(): Getting all values from EEPROM
[   191][E][Preferences.cpp:483] getString(): nvs_get_str len fail: WifiSSID NOT                                                       _FOUND
[   199][E][Preferences.cpp:483] getString(): nvs_get_str len fail: WifiPass NOT                                                       _FOUND
[   207][D][WifiMQTTManager.cpp:12] begin(): Wifi SSID: , Password:
[   215][E][Preferences.cpp:483] getString(): nvs_get_str len fail: MQTTServerIP                                                        NOT_FOUND
[   225][D][mEEPROM.cpp:73] putUInt(): PrefPutInt; 'MQTTPort' = '1883'
[   236][D][WifiMQTTManager.cpp:61] begin(): Wifi needs configuring, Starting AP                                                        Mode
[   254][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READ                                                       Y
[   339][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 10 - AP_START
[   354][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 11 - AP_STOP
[   361][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 10 - AP_START
[   402][D][WifiMQTTManager.cpp:69] begin(): DNS Server Started.
[  1409][D][HTTPWSFunctions.h:402] StartCriticalWebService(): Configuring Web Se                                                       rvices.
[  1417][D][HTTPWSFunctions.h:413] StartCriticalWebService(): Redirecting root t                                                       o index-ap.htm
Entering Configuration Mode Failure...
[  1637][E][CANBUS.cpp:57] Begin(): CAN Bus Failed to Initialise
[  1647][D][VeDirectFrameHandler.cpp:71] OpenSerial(): Opening Serial Port rxPin                                                       : 33, txPin 32
[ 16647][D][main.cpp:222] loop(): Send MQTT Param Update
[ 31653][D][main.cpp:222] loop(): Send MQTT Param Update
[ 46659][D][main.cpp:222] loop(): Send MQTT Param Update
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    10][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    53][D][main.cpp:105] setup(): EEPROM Store opened, initial key found.
E (55) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1367:error: Corrupted dir pair at {0x0, 0x1}

E (56) esp_littlefs: mount failed,  (-84)
E (58) esp_littlefs: Failed to initialize LittleFS
[    71][E][LittleFS.cpp:98] begin(): Mounting LittleFS failed! Error: -1
[... 732 lines cut ...]
[245955][E][vfs_api.cpp:105] open(): /littlefs/ipv6check.gz does not exist, no permits for creation
[245967][E][vfs_api.cpp:105] open(): /littlefs/ipv6check/index.htm does not exist, no permits for creation
[245978][E][vfs_api.cpp:105] open(): /littlefs/ipv6check/index.htm.gz does not exist, no permits for creation
[246208][E][vfs_api.cpp:105] open(): /littlefs/announce does not exist, no permits for creation
[246221][E][vfs_api.cpp:105] open(): /littlefs/announce.gz does not exist, no permits for creation
[246232][E][vfs_api.cpp:105] open(): /littlefs/announce/index.htm does not exist, no permits for creation
[246244][E][vfs_api.cpp:105] open(): /littlefs/announce/index.htm.gz does not exist, no permits for creation
[... cut ...]
[247562][E][vfs_api.cpp:105] open(): /littlefs/204/index.htm does not exist, no permits for creation
[247575][E][vfs_api.cpp:105] open(): /littlefs/204/index.htm.gz does not exist, no permits for creation
[
`

Ok, total rookie here, but above the dump from PuTTY.

I can see that when i click "Scan" for wifi, it triggers a "WiFiEvent(): [WiFi-event] event: 1", and right after, it drops the wifi connection (which i thought was a reboot previously).

After it drops the connection, i need to start over when it reconnects.

Any ideas?

sijones, 2024-09-28:
Can see in the log, it returns the results to your device then your device disconnects, strangely.

184502][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[184510][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 1

[188030][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 13 - AP_STADISCONNECTED
[188039][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 13

[188045][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[188053][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 12

[188059][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[188067][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 14

Try using a different device for setup, if you trying with a phone use a laptop or vice versa.

NeliusDK, 2024-09-29:
I tried phone and desktop with older Intel Wifi AC, both with the same symptoms where it disconnected as above.

I then tried my (very old), MS Surface Pro 3, and voila, it stayed connected and i could enter everything and reboot, all settings saved correctly.

I have no idea why the other computers would'nt stay connected, but it works now.
Thank you for your assistance.

closed 2024-09-29 by NeliusDK