sijones/DiyBatteryBMS #11: Setting ChargeEnable and DischargeEnable not working

NeliusDK, 2024-09-29:
I've updated the homeassistant yaml to control force charge, discharge and charge.

Force charge is working correctly.

Changing ChargeEnable and DischargeEnable have no effect.
My understanding is that when disabling charging, the charge limit should be set to 0 A. But it stays as originally confirgured when looking at the ESP, and therefore no changes on the inverter either. Same problem if you try to disable discharging.
Looking at the MQTT topics, the DischargeEnable and ChargeEnable under /Param, is correctly updated to ON or OFF.

The updated (topic equals the standard topic in the config, and it follows the new structure in HA) homeassistant yaml below.

`mqtt:
  # Home Assistant MQTT config for VE.DirectMQTTCANBUS, allows switch control of charge discharge
  # and force charging of the batteries - useful for off peak charging

  switch:
    - name: "PylonTech Protocol" # Choose an easy-to-recognize name
      unique_id: "PylontechProtocol"
      state_topic: "DIYBATTERYBMS/Param/EnablePYLONTECH" # Topic to read the current state
      command_topic: "DIYBATTERYBMS/set/EnablePYLONTECH" # Topic to publish commands
      qos: 1
      payload_on: "ON" # or "on", depending on your MQTT device
      payload_off: "OFF" # or "off", depending on your MQTT device
      retain: true # or false if you want to wait for changes

    - name: "Force Charge" # Choose an easy-to-recognize name
      unique_id: "ForceCharge"
      state_topic: "DIYBATTERYBMS/Param/ForceCharge" # Topic to read the current state
      command_topic: "DIYBATTERYBMS/set/ForceCharge" # Topic to publish commands
      qos: 1
      payload_on: "ON" # or "on", depending on your MQTT device
      payload_off: "OFF" # or "off", depending on your MQTT device
      retain: true # or false if you want to wait for changes

    - name: "Discharge Enable" # Choose an easy-to-recognize name
      unique_id: "DischargeEnable"
      state_topic: "DIYBATTERYBMS/Param/DischargeEnable" # Topic to read the current state
      command_topic: "DIYBATTERYBMS/set/DischargeEnable" # Topic to publish commands
      qos: 1
      payload_on: "ON" # or "on", depending on your MQTT device
      payload_off: "OFF" # or "off", depending on your MQTT device
      retain: true # or false if you want to wait for changes

    - name: "Charge Enable" # Choose an easy-to-recognize name
      unique_id: "ChargeEnable"
      state_topic: "DIYBATTERYBMS/Param/ChargeEnable" # Topic to read the current state
      command_topic: "DIYBATTERYBMS/set/ChargeEnable" # Topic to publish commands
      qos: 1
      payload_on: "ON" # or "on", depending on your MQTT device
      payload_off: "OFF" # or "off", depending on your MQTT device
      retain: true # or false if you want to wait for changes
`

sijones, 2024-09-29:
Make sure Pylontech protocol is off, and try again, the switch for pylontech changes certain behaviour for situations. So try that. You should see the charge / discharge current drop to 0.

NeliusDK, 2024-09-30:
Confirmed, disabling Pylontech protokol fixed it.
Thanks

closed 2024-09-30 by NeliusDK