sijones/DiyBatteryBMS #14: VEDisplay.cpp:15:17: error: no matching function for call to 'LCD_I2C::begin()

chomats, 2025-01-25:
I can build and I have an error : 

> `Processing esp32plus (platform: espressif32; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ ^2.1.3
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.1.4 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ ^3.2.2
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.3.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing heman/AsyncMqttClient-esphome @ ^2.1.0
Unpacking  [####################################]  100%
Library Manager: AsyncMqttClient-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing git+https://github.com/bblanchon/ArduinoJson.git
git version 2.48.1
Clonage dans '/home/jkgl0492/.platformio/.cache/tmp/pkg-installing-31zbyhjf'...
remote: Enumerating objects: 591, done.
remote: Counting objects: 100% (591/591), done.
remote: Compressing objects: 100% (464/464), done.
remote: Total 591 (delta 117), reused 244 (delta 61), pack-reused 0 (from 0)
Réception d'objets: 100% (591/591), 394.91 Kio | 4.59 Mio/s, fait.
Résolution des deltas: 100% (117/117), fait.
Library Manager: ArduinoJson@7.3.0+sha.8f7e793 has been installed!
Library Manager: Installing git+https://github.com/PaulStoffregen/Time.git
git version 2.48.1
Clonage dans '/home/jkgl0492/.platformio/.cache/tmp/pkg-installing-guljtsf5'...
remote: Enumerating objects: 38, done.
remote: Counting objects: 100% (38/38), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 38 (delta 2), reused 25 (delta 0), pack-reused 0 (from 0)
Réception d'objets: 100% (38/38), 26.01 Kio | 1.08 Mio/s, fait.
Résolution des deltas: 100% (2/2), fait.
Library Manager: Time@1.6.1+sha.a18e50d has been installed!
Library Manager: Installing git+https://github.com/coryjfowler/MCP_CAN_lib.git#1.5.0
git version 2.48.1
Clonage dans '/home/jkgl0492/.platformio/.cache/tmp/pkg-installing-4az9g5xc'...
remote: Enumerating objects: 33, done.
remote: Counting objects: 100% (33/33), done.
remote: Compressing objects: 100% (26/26), done.
remote: Total 33 (delta 2), reused 19 (delta 0), pack-reused 0 (from 0)
Réception d'objets: 100% (33/33), 42.17 Kio | 938.00 Kio/s, fait.
Résolution des deltas: 100% (2/2), fait.
Note : basculement sur '5439f064f4e6e06265407807bf87df401442cc3a'.

Vous êtes dans l'état « HEAD détachée ». Vous pouvez visiter, faire des modifications
expérimentales et les valider. Il vous suffit de faire un autre basculement pour
abandonner les commits que vous faites dans cet état sans impacter les autres branches

Si vous voulez créer une nouvelle branche pour conserver les commits que vous créez,
il vous suffit d'utiliser l'option -c de la commande switch comme ceci :

  git switch -c <nom-de-la-nouvelle-branche>

Ou annuler cette opération avec :

  git switch -

Désactivez ce conseil en renseignant la variable de configuration advice.detachedHead à false

Library Manager: mcp_can@1.5.0+sha.5439f06 has been installed!
Library Manager: Installing git+https://github.com/sijones/AsyncElegantOTA.git
git version 2.48.1
Clonage dans '/home/jkgl0492/.platformio/.cache/tmp/pkg-installing-6824_fap'...
remote: Enumerating objects: 24, done.
remote: Counting objects: 100% (24/24), done.
remote: Compressing objects: 100% (18/18), done.
remote: Total 24 (delta 0), reused 11 (delta 0), pack-reused 0 (from 0)
Réception d'objets: 100% (24/24), 407.01 Kio | 3.63 Mio/s, fait.
Library Manager: AsyncElegantOTA@2.2.10+sha.54dcc18 has been installed!
Library Manager: Installing git+https://github.com/hasenradball/LCD-I2C.git
git version 2.48.1
Clonage dans '/home/jkgl0492/.platformio/.cache/tmp/pkg-installing-8vbz9h2l'...
remote: Enumerating objects: 31, done.
remote: Counting objects: 100% (31/31), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 31 (delta 0), reused 16 (delta 0), pack-reused 0 (from 0)
Réception d'objets: 100% (31/31), 3.18 Mio | 13.72 Mio/s, fait.
Library Manager: LCD-I2C@1.0.1+sha.a614bcc has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- AsyncMqttClient-esphome @ 2.1.0
|-- ArduinoJson @ 7.3.0+sha.8f7e793
|-- Time @ 1.6.1+sha.a18e50d
|-- mcp_can @ 1.5.0+sha.5439f06
|-- AsyncElegantOTA @ 2.2.10+sha.54dcc18
|-- LCD-I2C @ 1.0.1+sha.a614bcc
|-- FS @ 2.0.0
|-- LittleFS @ 2.0.0
|-- WiFi @ 2.0.0
|-- SPI @ 2.0.0
|-- Preferences @ 2.0.0
|-- WebServer @ 2.0.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
Building in debug mode
Compiling .pio/build/esp32plus/src/CANBUS.cpp.o
Compiling .pio/build/esp32plus/src/VEDisplay.cpp.o
Compiling .pio/build/esp32plus/src/VeDirectFrameHandler.cpp.o
Compiling .pio/build/esp32plus/src/WifiMQTTManager.cpp.o
Compiling .pio/build/esp32plus/src/mEEPROM.cpp.o
Compiling .pio/build/esp32plus/src/main.cpp.o
Building .pio/build/esp32plus/bootloader.bin
Generating partitions .pio/build/esp32plus/partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully VEDisplay.cpp:15:17: error: no matching function for call to 'LCD_I2C::begin()'created esp32 image.
src/VEDisplay.cpp: In member function 'void Display::Begin(Display::DisplayType)':
src/
     _lcd->begin();
                 ^
In file included from src/VEDisplay.h:4,
                 from src/VEDisplay.cpp:1:
.pio/libdeps/esp32plus/LCD-I2C/src/LCD-I2C.h:49:14: note: candidate: 'void LCD_I2C::begin(TwoWire*)'
         void begin(TwoWire *wire);
              ^~~~~
.pio/libdeps/esp32plus/LCD-I2C/src/LCD-I2C.h:49:14: note:   candidate expects 1 argument, 0 provided
Compiling .pio/build/esp32plus/lib170/AsyncTCP-esphome/AsyncTCP.cpp.o
src/main.cpp:41:10: fatal error: VEDirectFrameHandler.h: No such file or directory

******************************************************************************
* Looking for VEDirectFrameHandler.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:VEDirectFrameHandler.h"
* Web  > https://registry.platformio.org/search?q=header:VEDirectFrameHandler.h
*
******************************************************************************

 #include "VEDirectFrameHandler.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/esp32plus/lib21c/FS/FS.cpp.o
*** [.pio/build/esp32plus/src/VEDisplay.cpp.o] Error 1
*** [.pio/build/esp32plus/src/main.cpp.o] Error 1
======================================================================== [FAILED] Took 10.90 seconds ========================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32plus      FAILED    00:00:10.899
=================================================================== 1 failed, 0 succeeded in 00:00:10.899 ===================================================================

 *  Arrêt du processus de terminal "platformio 'run', '--environment', 'esp32plus'". Code de sortie : 1. 
 *  Le terminal sera réutilisé par les tâches, appuyez sur une touche pour le fermer. 
`

tedohmatic, 2025-01-28:
_lcd->begin(); compile errors, the last version 2.02 of the DiyBatteryBMS code relied on a older version 0.3.0 of the LCD-I2C code: https://github.com/hasenradball/LCD-I2C/releases/tag/v0.3.0

To make DiyBatteryBMS build correctly, change the LCD-I2C version in the lib_deps section of platformio.ini for your board:

in platformio.ini replace:
https://github.com/hasenradball/LCD-I2C.git
with:
https://github.com/hasenradball/LCD-I2C.git#v0.3.0

Make sure to remove any cached version of the newer LCD-I2C or you will still have build problems

I'm on a mac so your path may vary, but remove the LCD-I2C containing the newer version:
~/YOUR_src/DiyBatteryBMS/.pio/libdeps/esp32.../LCD-I2C

you might want to make sure there aren't any download caches of the newer 1.0.0 or 1.0.1 version of LCD-I2C in
~/.platformio/.cache also

sijones, 2025-01-28:
I need to do a couple of changes and then a new release will come.

Please use the release rather than source code at the moment.

chomats, 2025-01-28:
Thanks

I see another problem : the file VeDirectFrameHandler.h is with a lower case and a upper case in the include referrence in 
main.cpp.
I compile on linux.

https://github.com/sijones/DiyBatteryBMS/commit/42a0487db242223ea17aa72d35c046741e20f89c

sijones, 2025-01-29:
@chomats @tedohmatic 

Source is updated with Lib fixes, using the latest LCD Library.

Smart adjust feature for battery charging isn't working, but I don't want to alter the code to disable it as i'll fix it soon.

tedohmatic, 2025-01-30:
Thanks @sijones , I rebuilt and installed, most things work, but mqtt isn't opening up outbound connections to the server.  LCD display shows MQTT: NO and everything else OK.  I'll try to do some debugging and give more details to help track.

sijones, 2025-01-30:
Make sure you're not trying to use mqtt secure, that isn't supported.

sijones, 2025-01-30:
The serial port has more info, so have the cable connected and serial
monitor open, reset device so you get the start up.

tedohmatic, 2025-01-31:
I checked with netstat on the mqtt server and the socket gets opened and receives MQTT when running the ESP32 with DIYBattery v 2.0.2, but when I upgrade to current main and set the MQTT server, it doesn't open the outbound connection to the server correctly.  I'll keep looking for the time being.

here is a recent serial output on new version:

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    36][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    95][D][main.cpp:120] setup(): EEPROM Store opened, initial key found.
[   103][D][main.cpp:121] setup(): EEPROM has free entries of: 412
[   110][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[   433][D][WifiMQTTManager.cpp:8] begin(): Getting all values from EEPROM
[   441][D][WifiMQTTManager.cpp:13] begin(): Wifi SSID: myssid, Password: `?
[   460][D][WifiMQTTManager.cpp:48] begin(): Connecting to SSID: myssid
[   488][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   570][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[   596][D][WifiMQTTManager.cpp:53] begin(): Setting up mDNS Service.
[   622][D][mqttFunctions.h:213] mqttsetup(): Using IP Address method to connect.
Entering Configuration Mode Successful!
Setting Baudrate Successful!
[   640][I][CANBUS.cpp:50] Begin(): CAN Bus initialised
[   642][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   657][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 4

[   667][I][CANBUS.cpp:21] canSendTask(): Starting CAN Bus send task
[   683][D][VeDirectFrameHandler.cpp:71] OpenSerial(): Opening Serial Port rxPin: 33, txPin 32
[   696][D][HTTPWSFunctions.h:9] TaskSetClock(): Entering TaskSetClock
[   703][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   704][E][Preferences.cpp:483] getString(): nvs_get_str len fail: NTPServer NOT_FOUND
[   718][D][HTTPWSFunctions.h:14] TaskSetClock(): No NTP Server Set.
[   718][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 192.168.11.52, MASK: 255.255.255.0, GW: 192.168.11.1
[   735][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 7

[   741][D][mqttFunctions.h:103] WiFiEvent(): WiFi connected, IP Address: 192.168.11.52
[   749][I][mqttFunctions.h:91] connectToMqtt(): Connecting to MQTT...
[   756][I][mqttFunctions.h:93] connectToMqtt(): Using User: , Password: 
[   771][E][CANBUS.cpp:385] SendCANData(): Failed to Send CAN Packets: 6
[   778][E][CANBUS.cpp:25] canSendTask(): Failure returned from SendAllUpdates
[   778][D][mqttFunctions.h:132] onMqttDisconnect(): Disconnected from MQTT.
[   782][D][HTTPWSFunctions.h:590] taskStartWebServices(): Starting Web Services
[   799][D][HTTPWSFunctions.h:509] StartWebServices(): Configuring Web Services.
[   806][D][HTTPWSFunctions.h:523] StartWebServices(): Redirecting root to index.htm
[  1028][D][VeDirectFrameHandler.cpp:261] frameEndEvent(): Frame Index Value 16
[  1035][D][main.cpp:230] loop(): Data Available to Process
[  1502][D][VeDirectFrameHandler.cpp:261] frameEndEvent(): Frame Index Value 12
[  1510][D][main.cpp:230] loop(): Data Available to Process
[  1516][I][DataProcessing.h:26] VEDataProcess(): Battery Voltage Update: 13243V
[  1523][I][DataProcessing.h:37] VEDataProcess(): Battery Current Update: 0mA
[  1530][I][DataProcessing.h:49] VEDataProcess(): Battery SOC Update: 1000%
[  1799][D][CANBUS.cpp:411] AllReady(): All initial and inverter data set, going to running mode.
[  1811][D][main.cpp:247] loop(): Send MQTT Param Update
[  1851][E][CANBUS.cpp:385] SendCANData(): Failed to Send CAN Packets: 6
[  1857][E][CANBUS.cpp:25] canSendTask(): Failure returned from SendAllUpdates
[  2002][D][VeDirectFrameHandler.cpp:261] frameEndEvent(): Frame Index Value 16
[  2010][D][main.cpp:230] loop(): Data Available to Process
[  2015][I][DataProcessing.h:26] VEDataProcess(): Battery Voltage Update: 13243V
[  2023][I][DataProcessing.h:37] VEDataProcess(): Battery Current Update: 0mA
[  2030][I][DataProcessing.h:49] VEDataProcess(): Battery SOC Update: 1000%
[  2502][D][VeDirectFrameHandler.cpp:261] frameEndEvent(): Frame Index Value 12
[  2510][D][main.cpp:230] loop(): Data Available to Process
[  2516][I][DataProcessing.h:26] VEDataProcess(): Battery Voltage Update: 13244V
[  2523][I][DataProcessing.h:37] VEDataProcess(): Battery Current Update: 0mA
[  2530][I][DataProcessing.h:49] VEDataProcess(): Battery SOC Update: 1000%
[  2798][I][mqttFunctions.h:91] connectToMqtt(): Connecting to MQTT...
[  2805][I][mqttFunctions.h:93] connectToMqtt(): Using User: , Password: 
[  2822][D][mqttFunctions.h:132] onMqttDisconnect(): Disconnected from MQTT.

sijones, 2025-01-31:
I'll add some more logging to the MQTT connect/disconnect method, it tries and fails to connect so looks like it will be better if I see if I can log the reason it did/didn't connect, and the reason for disconnection, as always it works for me! This source code is running now in my system.

tedohmatic, 2025-01-31:
I'll keep hacking and appreciate your view on it too.  It's a good opportunity to dive into the code, but free time is at a premium lately for me.  I'll send an update to diysolar when I have some new pics for the hardware assembly I made over the past couple of weeks.

sijones, 2025-02-01:
I've added more logging for the MQTT disconnect, so should be able to see the reason in the log's.

tedohmatic, 2025-02-01:
Thanks!  I added more logging, it seems the mqttport isn't getting read from prefs right and then defaults to '0', even though it is set as 1883 in the web interface.

```
[   635][D][mqttFunctions.h:246] mqttsetup(): Using IP Address method to connect.
[   643][D][mqttFunctions.h:253] mqttsetup(): MQTT Enabled: True
[   649][D][mqttFunctions.h:254] mqttsetup(): MQTT Subscribe: 
[   654][D][mqttFunctions.h:255] mqttsetup(): MQTT Server: 10.252.0.50
[   659][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   668][D][mqttFunctions.h:100] WiFiEvent(): [WiFi-event] event: 4
[   675][D][mqttFunctions.h:256] mqttsetup(): MQTT Port: 0
[   680][D][mqttFunctions.h:257] mqttsetup(): MQTT User: 
[   685][D][mqttFunctions.h:258] mqttsetup(): MQTT Pass: 
[   691][D][mqttFunctions.h:259] mqttsetup(): MQTT Topic: DIYBATTERY2
[   697][D][mqttFunctions.h:260] mqttsetup(): MQTT Data Topic: DIYBATTERY2/Data
```

I hardcoded 1883 as default in mqttfunctions.h until I can figure out if the pref is not getting stored or read correctly.

    //uint16_t mqttPort = (uint16_t) pref.getUInt16(ccMQTTPort,mqttPort);
    uint16_t mqttPort = (uint16_t) pref.getUInt16(ccMQTTPort, 1883);

sijones, 2025-02-02:
Try a flash erase and then flash again, only thing I can think of is 0 is stored but that should of shown in web browser.

tedohmatic, 2025-02-02:
This is really strange.  I tried it on two separate ESP32 boards, a 30 pin and a 38 pin devboard.  Both exhibited the same behavior with 0 being set at reboot.  I erased both and flashed and went through the littlefs OTA update and repopulated the settings.  The port 1883 is indeed set in the web interface and it also persists between power cycles in the Wifi/MQTT web settings page.  Though the logging shows 0 as the mqtt port and same TCP Disconnect error.

sijones, 2025-02-03:
ok, i'll do some testing on a fresh device.

I've had weirdness like this when the Libs are not quite right, it's usually the async TCP lib that's a bit of a problem, Async MQTT uses it but the original lib is old and isn't updated, it's been substituted by a newer one that was fine, but possibly something has changed and broken something.

closed 2026-02-13 by sijones